Gepetto/example-robot-data #1: Renaming the project

jcarpent, 2019-08-29:
@nim65s I think a better name for the project is expected for clarity and self-explainability.
`many-robot-descriptions` may be a candidate.

nim65s, 2019-08-29:
@cmastalli what do you think about this ?

cmastalli, 2019-08-29:
To be honest, I don't like neither the keyword "data", instead using description is a good idea seems most people are familiar with this because ROS community. What about just description (without plural)? Instead we could might put the plural in robot, i.e. "robots".

What it doesn't convince me yet is the keyword "many". I remember that @ostasse likes the keyword "example". His point is something like this "I don't want people use these URDFs for our real robots, then using example make clearly state this". @ostasse also mentioned that this is important for our partners (PAL robotics).

So probably we could rename the repository to something like: "example-robots-description" (or shorter "erobots-description"). Personally, I like the second one we can always add a good self-explanatory description of this package.

nim65s, 2019-08-29:
"erobots" could be understood as "e-robot", which doesn't mean much and would therefore make a good name for science fiction films, or "ero-bots", which mean something else we don't want to see here…

The contraction "e(xample-)robots" is not clear for me.

example-robots-descriptions is fine for me. The 2 plurals can explain both that we have multiple robots and that we have multiple descriptions for some of them, and could replace the "many" suggested by @jcarpent .

nim65s, 2019-08-29:
but another version of example-robot(s)-description(s) is also good for me

jcarpent, 2019-08-29:
robot-description-collection

nim65s, 2019-08-29:
I agree with @olivier-stasse, it has to be clear that these must not be used on real robots, and they can only serve in examples, unit-tests or benchmarks. Maybe "example" is not the best word for that, but this idea has to be in the title

jcarpent, 2020-08-19:
Weird naming, but lets keep it as people have become familiar to it.

closed 2020-08-19 by jcarpent