Gepetto/example-robot-data #12: missing files for FANUC robots

dadouz, 2020-01-22:
i want to ask you if you you can add a folder with related files for following robots: 
- Fanuc R-2000iB 165F 
- Fanuc CR35iA

we want to use the examples from pinocchio.

wxmerkt, 2020-01-22:
I am not sure we want to add all possible robots to the example repository. You can easily write a new robot wrapper for existing URDF/SRDF pairs.

gabrielebndn, 2020-01-22:
Hi @dadouz, this repo is mostly conceived as a set of shortcuts for the stack of tasks development team. Its use in real projects in not really recommended.
Pinocchio can perfectly work (and usually works) without this repo and you should not let yourself be blocked by the absence of a certain model here.
You can find at [pinocchio/examples/robot-wrapper-viewer.py](https://github.com/stack-of-tasks/pinocchio/blob/master/examples/robot-wrapper-viewer.py) an example on how to use `RobotWrapper` for any URDF, without the help of this package.
Further, at [pinocchio/examples/geometry-models.py](https://github.com/stack-of-tasks/pinocchio/blob/master/examples/geometry-models.py), you can find a way to achieve the same results *without* `RobotWrapper` (`RobotWrapper` itself is only a facilitation for newcomers and is not necessary at all for using Pinocchio).

jcarpent, 2020-08-19:
I would be in favour of supporting a maximum number of robots.

jcarpent, 2021-03-17:
@dadouz Could you add the robot inside this repo, it should be relatively easy.

cmastalli, 2021-03-17:
I would also be in favored to add these two robots.
But remember that the mission of this package is to provide a set of robots for prototyping, unittesting and benchmarking.
Its intension is not to be used for real applications on robots.

@dadouz we are not planning to include ourself these robots. However, you're welcome to contribute on this topic.

Let us know if you are planning to integrate these robots, otherwise, I would kindly ask you to close this issue :)