Gepetto/example-robot-data #16: __main__.py should check for gepetto-gui

nim65s, 2020-04-02:
And if gepetto-gui is not found, it should launch it.

But this behavior could be implemented at the gepetto-viewer / gepetto-viewer-corba level.

Maybe we should consider other viewers too…

nim65s, 2020-04-02:
ref https://github.com/humanoid-path-planner/hpp-rbprm-corba/blob/devel/src/hpp/corbaserver/rbprm/__main__.py#L51

wxmerkt, 2020-04-26:
Not sure we want this in this repository? I think it would be better to handle this at the gepetto-viewer level to avoid adding an assumption and dependency on a package providing URDFs+meshes.

cmastalli, 2020-04-26:
What do you mean with check? 

Additionally this repository uses Gepetto viewer for quick visualization to interested users. I don't think we need to support other viewers (just to make easier maintenance).

nim65s, 2020-04-27:
@wxmerkt : I think we should add this in gepetto-viewer, and then test that it works well here.

@cmastalli : I mean

```
╰─>$ python -m example_robot_data solo
/usr/lib/python3.8/site-packages/pinocchio/robot_wrapper.py:230: UserWarning: Error while starting the viewer client.
Check whether gepetto-viewer is properly started
  self.viz.initViewer(*args, **kwargs)
```

This could launch gepetto-gui, instead of showing an error that doesn't actually display the right name of the binary the user should launch.

hjw-1014, 2021-05-17:
Hi, @nim65s . I also wanna check how to launch gepetto-gui. I tried the same thing as you mentioned:

```
huangjiawei@huangjiawei-Ubuntu:/opt/openrobots/share$ python3 -m example_robot_data solo
/opt/openrobots/lib/python3.6/site-packages/pinocchio/robot_wrapper.py:272: UserWarning: Error while starting the viewer client.
Check whether gepetto-viewer is properly started
  self.viz.initViewer(*args, **kwargs)
```

Cam this be solved? Thanks in advance.