Gepetto/example-robot-data #30: INSTALL_PYTHON_INTERFACE_ONLY not the right naming

jcarpent, 2020-06-19:
It would be better to have a variable called:
`INSTALL_ROBOTS_DESCRIPTIONS`

cmastalli, 2020-06-19:
Could you elaborate your suggestion?
Is this related with the need of setting the `ROS_PACKAGE_PATH` variable?

jcarpent, 2020-06-19:
It is not clear what is not included when activating `INSTALL_PYTHON_INTERFACE_ONLY`. It would be better to set or unset something related to the installation of the meshes.

nim65s, 2020-06-19:
@cmastalli : "installing only the python module" == "not (installing the robot descriptions)". Almost equals, because we also have tons of generated stuff and metadata, like *.pc, *.cmake, some *.h, documentation, and package.xml.

@jcarpent : Do you need this variable ? is it raising any issue ?

I created this variable to have one easy way to split projects between python modules and the other things. This is the way to provide packages that can be used without the python modules, or with different version of pythons.

This variable with this name was made for this purpose, I think the name is explicit enough, and it fits my need. Plus this is the exact same name as in many other packages, so it's easier to factorize packaging tools.

nim65s, 2020-06-19:
@jcarpent : if you need a way to activate or deactivate installation of the meshes, we can do that, but it is almost not related with `INSTALL_PYTHON_INTERFACE_ONLY`.
We could say that `INSTALL_PYTHON_INTERFACE_ONLY` implies not installing meshes, but I don't think this statement would be useful.

cmastalli, 2020-06-19:
Maybe it would be a more clear the name `INSTALL_ONLY_PYTHON_MODULE`. However, I am fine with the actual name.