Gepetto/example-robot-data #35: modelPath not pointing to the right location

jcarpent, 2020-06-19:
@nim65s @cmastalli 

I think there is a current issue in the way the Python libs is retrieving the absolute mesh PATH.
Currently, all the meshes are related to `package:\\example-robot-data/robots/foo_description` but when looking for the mesh path, this one is pointing to:
`${CMAKE_INSTALL_PREFIX}/example-robot-data/robots`, meaning that there is an `example-robot-data/robots` which is duplicated.

nim65s, 2020-06-19:
ref mainly #19, but also #9 & #21

nim65s, 2020-06-19:
On my side, 
```python
>>> example_robot_data.getModelPath('/panda_description/urdf/panda.urdf')
'/home/gsaurel/local/robotpkg/install/share/example-robot-data/robots'
````

So I think this is the intended behavior.

jcarpent, 2020-06-19:
But it the path for the visuals should be: `/home/gsaurel/local/robotpkg/install/share`

nim65s, 2020-06-19:
fixed in 6a1807747f58ba467c67e017f4056451ec12efb8 ; @wxmerkt @cmastalli : can you test that please ?

nim65s, 2020-06-19:
I guess the list implies that we could provide the old one and the new, just to be sure… But this is cluttered enough already, so if we can avoid that it's better.

jcarpent, 2020-06-19:
Thanks a lot @nim65s for your precious help.

wxmerkt, 2020-06-19:
From reading the commit, it looks good (didn't run it locally)

closed 2020-06-20 by nim65s