Gepetto/example-robot-data #50: Talos collision shape of feet is different than in talos_data

pFernbach, 2020-09-17:
See https://github.com/Gepetto/example-robot-data/blob/a9dea8c2dfe28ae5bf36f1e26a7936fdc49be440/robots/talos_data/robots/talos_reduced.urdf#L2187-L2201  and  https://gitlab.laas.fr/stack-of-tasks/talos-data/-/blob/master/urdf/talos_reduced.urdf#L2187-2209

The model of talos_data is the correct one, the box shape used in example-robot-data have two issue: It is lower than the real feet (~5mm closer to the ground) and it do not represent well the collisions around the ankle.

cmastalli, 2020-09-17:
From my personal experience, it is better (faster and more accurate) to define a box instead of using a mesh file.

I understand that our box is not right, but I would vote to just define the proper values rather than copy pasting the talos-data one.

jcarpent, 2020-09-17:
I would say, we should use the initial model from PAL if we want to avoid further burden at the user level. The change are not documented and it might imply that bugs are occuring because of this.

cmastalli, 2020-09-17:
> I would say, we should use the initial model from PAL if we want to avoid further burden at the user level. The change are not documented and it might imply that bugs are occuring because of this.

This code stars from talos-data which this latter one starts from Pal code. I believe this error might be existed in root repos

pFernbach, 2020-09-18:
> I would say, we should use the initial model from PAL if we want to avoid further burden at the user level. The change are not documented and it might imply that bugs are occuring because of this.

Yes, this already caused such issue. 

Both version comes from PAL and depend on an argument that you must specify when building the urdf from the xacro : 
https://gitlab.laas.fr/stack-of-tasks/talos-data/-/blob/master/robots/talos_lower_body.urdf.xacro#L18

talos_data use 'default' and example-robot-data use 'thinbox'. This last one is absolutely not accurate and will not detect a lot of collisions (especially self collision when doing turns).

jcarpent, 2020-09-18:
@pFernbach I would suggest fixing this inaccuracy. Could you handle it? Maybe, it would be nice to have both models available. The default one being the most accurate one.

pFernbach, 2020-09-18:
I can. This mean using the mesh in the `talos_reduced.urdf` and creating a new urdf with the box ? `talos_reduced_box.urdf` ?

jcarpent, 2020-09-18:
Yes, this is the idea. Thanks.

jcarpent, 2020-10-05:
Solved by #51.

closed 2020-10-05 by jcarpent