Gepetto/example-robot-data #57: Uniform API of the loaders

proyan, 2020-10-28:
Hi guys,

The API of the robot loaders is too confusing. For example, to load solo, the API: `loadSolo()`, and to load solo12, API: `loadSolo(False)`

The reason we have the `load` functions, is to be simplify the loading of the urdfs/srdfs by just one single command. In order to do that, the `load<robotName>` the `robotName` here should be clear enough.

However, what I often have to do is to look at the implementation inside the `load...` function to see what is being loaded with which argument. Or I have to look at the arguments to predict what is being loaded where.

For example, consider `loadTalos(legs=False, arm=False, full=False, box=False)`
Here, we have 2**4=16 possible arguments. However, only some of these arguments are valid, and the rest are non-sense. If I want to load the legs of Talos, I need to predict which specific argument to give. Or I need to understand that full=True doesn't equate to arm=True  + legs=True. Finally, I end up looking at the implementation each time to make sure that I'm loading the right thing.

If eventually someone has to go through the implementation, he/she would directly look at the implementation of RobotWrapper and then we don't need these _easy to use_ loaders anymore.

What was the reason to deprecate loadTalosArms or loadTalosLegs?

nim65s, 2020-10-28:
The idea is to have only one list of available robots. For now, this list consists of strings. It would be better to use an Enum, but for now, you are expected to use:
`load('solo')` / `load('solo12')` / `load('talos')` / `load('talos_arm')` / `load('talos_legs')`.

As this list is used in many places in the code base, it is important to have it only once, to avoid duplication. Right now, it is in the keys of the `ROBOTS` dict, and the values of this dictionary allows `load()` to find the right function and arguments to use.

proyan, 2020-10-28:
I see. I agree `load(<robotname>)` is a much better option than what I'm using.
In that case, we should be deprecating `load<robotName>(*args)` to make users use load() as the only available choice. I have to confess, I did not know `load()` existed untill 7 minutes ago

nim65s, 2020-10-28:
I agree, but technically we can't deprecate `load<robot>()`, because, at least for now, `load(<robot>)` is mostly launching `load<robot>()`

nim65s, 2020-10-28:
Or we would have to rename `load<robot>()` into an internal `_load<robot>()`, and create `load<robot>()` that are deprecated and launch `_load<robot>()` and make `load(<robot>)` also load `_load<robot>()`. Which is slighly boring, but I could do that.

cmastalli, 2020-10-28:
> Or we would have to rename `load<robot>()` into an internal `_load<robot>()`, and create `load<robot>()` that are deprecated and launch `_load<robot>()` and make `load(<robot>)` also load `_load<robot>()`. Which is slighly boring, but I could do that.

This will be the right way, but I agree it is very boring task if we don't want to break somebody else code.

PR #58 (2020-11-24): RobotLoader
Hi,

This is a suggestion following discussions in #56.

With this, the main entry point of the public API becomes either:

```python
robot = load('talos')
```

or, if someone wants more that just the robot wrapper,

```python
robot, q0, urdf, srdf = load_full('talos')
```

This suggestion:
- make the code more generic
- avoid multiple calls to `getModelPath()` (which would be necessary for https://github.com/Gepetto/example-robot-data/pull/56#issuecomment-717980586)
- fix #57 
- is open for comments
closed 2021-03-10 by nim65s