Gepetto/example-robot-data #67: Talos not correct for pybullet - No link should be left without inertia

olivier-stasse, 2021-03-03:
In pybullet, links should have an inertia and a mass specified even if this zero.
Otherwise the default mass value is 1 Kg.
See https://github.com/robotology/icub-models/issues/12

A PR follows fixing this issue.

jcarpent, 2021-03-03:
This is awesome!

olivier-stasse, 2021-03-03:
What I should have mentionned is that I did the corrections for TALOS but not for the other robots...

olivier-stasse, 2021-03-03:
It looks like that at least the following files would be affected:
 * kinova.urdf < link name="j2s6s200_end_effector" / >
 * ur5_joint_limited_robot.urdf:  < link name="tool0" / >
 * ur3_robot.urdf:  < link name="tool0" / >
 * ur10_robot.urdf:  < link name="tool0" / >
 * ur3_joint_limited_robot.urdf:  < link name="tool0" / >
 * romeo.urdf:  < link name="l_sole" / >
 * tiago.urdf:  < link name="hand_grasping_frame" / >
 * tiago_no_hand.urdf:  < link name="base_footprint" / >
 * romeo_laas_small.urdf:    < link name="AccelerometerDownTorso_sensor" / >
 * tiago.urdf:  < link name="base_footprint" / >

wxmerkt, 2021-03-03:
I performed those fixes for Anymal a while back - but agree we should carry them through all URDFs. I recall at some point some dynamics libraries or simulators did not like "0" inertia either (i.e. crashed) so we went with 1e-6 or 1e-9. I can't recall which where affected though.

PR #68 (2021-03-10): [talos_reduced] Fix inertia and mass to zero for optical links.
Fix #67
closed 2021-03-10 by nim65s