Gepetto/example-robot-data #75: parsing error while loading universal robot urdf/dae files

ntorresalberto, 2021-04-16:
Hi,
From what I read [here](https://github.com/stack-of-tasks/pinocchio/issues/838) it could be related to the hpp-fcl library.

Basically I'm trying this:

```python
import example_robot_data
robot = example_robot_data.load('ur10')
```

```bash
$ python -i mwe_load_ur10.py 
Traceback (most recent call last):
  File "mwe_load_ur10.py", line 2, in <module>
    robot = example_robot_data.load('ur10')
  File "/home/user/.conda/envs/pinocchio/lib/python3.8/site-packages/example_robot_data/robots_loader.py", line 506, in load
    return loader(name, display, rootNodeName).robot
  File "/home/user/.conda/envs/pinocchio/lib/python3.8/site-packages/example_robot_data/robots_loader.py", line 493, in loader
    inst = ROBOTS[name]()
  File "/home/user/.conda/envs/pinocchio/lib/python3.8/site-packages/example_robot_data/robots_loader.py", line 64, in __init__
    self.robot = RobotWrapper.BuildFromURDF(self.urdf_path, [join(self.model_path, '../..')],
  File "/home/user/.conda/envs/pinocchio/lib/python3.8/site-packages/pinocchio/robot_wrapper.py", line 17, in BuildFromURDF
    robot.initFromURDF(filename, package_dirs, root_joint, verbose, meshLoader)
  File "/home/user/.conda/envs/pinocchio/lib/python3.8/site-packages/pinocchio/robot_wrapper.py", line 21, in initFromURDF
    model, collision_model, visual_model = buildModelsFromUrdf(filename, package_dirs, root_joint, verbose, meshLoader)
  File "/home/user/.conda/envs/pinocchio/lib/python3.8/site-packages/pinocchio/shortcuts.py", line 45, in buildModelsFromUrdf
    geom_model = pin.buildGeomFromUrdf(model, filename, geometry_type, package_dirs)
ValueError: Mesh index: 0
Face index: 0
```

I can reproduce the same error while using pinocchio with [ros-industrial](https://github.com/ros-industrial/universal_robot) while trying to `buildFromUrdf` to load ur3, ur3e, ur5e, ur10, ur10e.

Some info about my setup: I'm running Fedora 32 and all libraries were installed from conda-forge.

jcarpent, 2021-04-16:
This issue is due to Assimp in fact. I will try to fix it by regenerating correctly the mesh.

jcarpent, 2021-04-16:
But `robot = example_robot_data.load('ur5')` works well.

ntorresalberto, 2021-04-16:
> But `robot = example_robot_data.load('ur5')` works well.

Yes, exaclty.
I listed the UR robots that cause this problem with `universal-robots`, I don't know if it's useful but ur5 works and ur5e doesn't.

> This issue is due to Assimp in fact. I will try to fix it by regenerating correctly the mesh.

Thank you.
I don't know what that is, but let me know if I can be of help.

cmastalli, 2021-04-16:
@jcarpent -- we need to update our unittest as well, could you handle it?

Unfortunately, we are just testing UR5 (see https://github.com/Gepetto/example-robot-data/blob/master/unittest/test_load.py#L106-L110)

jcarpent, 2021-04-16:
I'm adjusting the policy inside hpp-fcl. Please see https://github.com/humanoid-path-planner/hpp-fcl
@cmastalli I don't have time to update the tests. I let you do this task.

cmastalli, 2021-04-16:
I have created a PR with the need unittests. However, we need to wait a new HPP-FCL release before merging it.
@jcarpent -- could we possible to have a soon release in HPP-FCL?

cmastalli, 2021-05-04:
This has been solved!

closed 2021-05-04 by cmastalli