Gepetto/example-robot-data #82: Is tiago_dual_robot.urdf provided?

hjw-1014, 2021-05-19:
Hi, I'm very new to urdf. I need a **TIAGo++** file, eg. tiago_dual_robot. 

Tiago robot (**tiago.urdf**) is provided here, but it has only one hand, so I wanna ask, Is there also **tiago_dual_robot.urdf** provided or should I edit urdf of **tiado.urdf** here to add an arm?

I found[ here ](https://github.com/pal-robotics/tiago_robot), tiago_dual_description is provided, but inside is xacro file. 

Thanks in advance.

nim65s, 2021-05-19:
Hi,

I think I can add this robot here. I'll give it a try tomorrow.

hjw-1014, 2021-05-19:
> Hi,
> 
> I think I can add this robot here. I'll give it a try tomorrow.

Thank you so much, that would help me a lot.

commit f851c4be (2021-05-26): Merge pull request #83 from nim65s/topic/tiago_dual

Add tiago dual, fix #82
closed 2021-05-26 by nim65s