Gepetto/example-robot-data #87: question about mass distribution in double pendulum

nmansard, 2021-06-03:
The double pendulum has been introduced by @PepMS so I guess this question is mostly for him, yet I open it here if anybody else can suggest relevant remark.
The masses chosen for the double pendulum are
```
[ m.mass for m in r.model.inertias]                                                                                                 
[0.10159, 0.26703, 0.33238]
```
Is there a reason why the mass of the second body (0.33) is larger than the mass of the first body (0.26)? Where are these values coming from?

Thx

PepMS, 2021-06-03:
There's no specific reason. This was meant to be a toy example that helped me to enter the Crocoddyl world. I did this rough design in CAD and assigned materials to every part (I chose steel for no specific reason), hence the values. 

I kind of remember that with a larger mass for the second body the FDDP solver converged faster from a cold-start. Probably, by designing a better cost function I would've had better results too. But did not dig much into it since, as I said, this was an introductory exercise for me.

From my side, feel free to change anything you need. Also, if you'd like something more realistic or precise I'll be glad to help.

nmansard, 2021-06-03:
ok thx.

closed 2021-06-03 by nmansard