Gepetto/example-robot-data #90: Add FingerEdu robot

jviereck, 2021-06-12:
If possible, I would like to add the FingerEdu to the example-robot-data package. We use this 3-dim fixed-base robot as educational platform for teaching. If the students can pip/conda-install the robot, it would make our life much easier.

If you are okay to add it I can try to make a PR.

jviereck, 2021-06-12:
Here is the link to the robot: https://github.com/open-dynamic-robot-initiative/open_robot_actuator_hardware/blob/master/mechanics/finger_edu_v1/README.md#fingeredu-v1

[image]

jcarpent, 2021-06-13:
Yes, sure! This is more than welcome. You can proceed with a new PR. Thanks a lot @jviereck for this suggestion.

closed 2021-06-15 by nim65s