Gepetto/example-robot-data #92: Rename Solo to Solo8

jviereck, 2021-06-14:
There is a slight naming error right now in the package. When calling `load("solo")` it creates a Solo8. Note however, that "Solo" is the name of the quadruped platform as such, not of any specific robot.

Therefore, it would be better to have `load("solo8")`. Then, there are two versions of solo8: One with TI cards (called `solo8v1`) and one with udrivers (called `solo8v2`). They have different weights. It looks like the one in the repo is from the `solo8v2` model.

Therefore, would it make sense to make `load("solo")` become `load("solo8v2")`?

nim65s, 2021-06-15:
For the first part, yes, we can rename "solo" into "solo8". I think we'll just need to keep "solo" working for a while, but with a deprecation warning.

For the second part, I'm not sure to understand your needs. For now, this project aims "to be used in benchmarking, unit-tests, teachings, tutorials or show-cases" ; but in those use cases, it doesn't really matter if there is a particular controller or another, and if the weight is changed a bit.

And for other robots, we are not always including all options regarding different lidars / grippers / IMUs etc. But maybe we need to discuss that, I don't know.

jviereck, 2021-06-15:
Thanks @nim65s for your input. Based on it I went with the name "Solo8" for now.  I created PR #93 for this.

closed 2021-06-17 by nim65s

jviereck, 2021-06-17:
Thanks for the merge @nim65s . 

Would it be possible to make a new release that include the finger_edu and bolt?

commit 475f4997 (2021-06-17): Rename "Solo" to "Solo8". Fixes #92
commit 5d6f078d (2021-06-17): Merge pull request #93 from jviereck/jviereck/solo8

Rename "Solo" to "Solo8". Fixes #92