Gepetto/example-robot-data #97: [question] bolt initial configuration

gfadini, 2021-06-25:
Hi,
I'm not sure if here is the right place to ask, maybe I should replicate over to https://github.com/open-dynamic-robot-initiative/robot_properties_bolt
I noticed that there is an small gap ~3cm between the foot and the ground for the bolt model for the initial configuration in the srdf.
Is this done on purpose to include the dimensions of the foot which is missing in the model?
If not, maybe the offset should be removed in ```/robots/bolt_description/srdf/bolt.srdf``` line 25 with the change
```
-        <joint name="root_joint" value="0. 0. 0.35487417 0. 0. 0. 1." />
+        <joint name="root_joint" value="0. 0. 0.32478765 0. 0. 0. 1." />
```

jcarpent, 2021-11-04:
I do think you should first discuss this issue with the creators of BOLT ;)

gfadini, 2021-12-02:
You're right @jcarpent, these details really depend on how the robot models were generated, I'll close the issue

closed 2021-12-02 by gfadini

cmastalli, 2021-12-02:
I think we could integrate the suggested modifications in this repo.

In the other robots, we have not such kind of offset. @gfadini feel free to make a PR here.