Gepetto/example-robot-data #99: Relicense or Remove Tiago

nim65s, 2021-07-22:
As found by #98, Tiago is licensed by a [Creative Commons BY-NC-ND 3.0](https://github.com/pal-robotics/tiago_robot/blob/kinetic-devel/tiago_description/package.xml#L12), which clearly forbids us to distribute this derivated work.

We should either :
- kindly ask PAL-Robotics for either a reconsideration of the license of this package, or to explicitely give us an exception
- remove Tiago from example-robot-data

cmastalli, 2021-07-23:
I believe we should ask PAL first and explain them our mistake. Otherwise, we have to remove it.

Removing the Tiago robot might repercute many folks. So, if needed, we should print a deprecated message explaining the major reasons for the remotion.

jcarpent, 2021-09-17:
Tiago and Talos do not have the same license?

nim65s, 2021-09-17:
no

nim65s, 2021-11-22:
I got an oral confirmation from PAL CEO & CTO: we can keep working like that. They will fix the license on their side when they will have time.

closed 2021-11-22 by nim65s