Gepetto/example-robot-data #112: Moving simple_humanoid from pinocchio/models to example_robot_data

proyan, 2021-11-30:
I suggest that we move pinocchio's `simple_humanoid` from `pinocchio/models` to `example_robot_data`. It would make the sample humanoid available for other packages as well. Any views?

jcarpent, 2021-11-30:
Totally fine on my side.

proyan, 2021-11-30:
Ah my bad! It looks like it is already there, but is duplicated in pinocchio! This issue is ill-placed, I'll close it.

proyan, 2021-11-30:
I'll create a PR synchronizing the models from pinocchio and here, let's change the direction of this issue :)

cmastalli, 2021-11-30:
It sounds good for me as well.
Thanks @proyan for taking care of it :)

cmastalli, 2022-03-31:
Is this issue sorted out?

nim65s, 2022-04-01:
Actually, I added this robot in #69,  8 months before this PR, so I'm not sure what this PR is about.

proyan, 2022-04-01:
#120 handles the changes I was proposing