Gepetto/example-robot-data #159: Error when loading urdfs to ROS parameter sever

james-p-foster, 2023-02-15:
Hello,

I've recently been working with some of these urdfs alongside ROS. A typical use case of mine when replaying a rosbag is to:
1. Activate `roscore` and `rviz`
2. Play the rosbag
3. Add the urdf to the ROS parameter server via: `rosparam load <path-to-urdf> robot_description` so that I can view the robot urdf in `rviz`.

On that final third step, `rosparam load` results in a `pyyaml` `Scanner Error` for some of the urdfs.  From what I gather, the urdfs are being read as yaml files, and the scanner is failing to read colons followed by whitespace, e.g. `: `. I believe this is due to the yaml scanner reading the colon and whitespace as indicating a key-value pair (see e.g. [here](https://stackoverflow.com/questions/10971621/error-parsing-yaml-file-mapping-values-are-not-allowed-here)). An example error from `anymal_b_simple_description`:
```
yaml.scanner.ScannerError: mapping values are not allowed here
  in "<unicode string>", line 288, column 24:
          <!-- for a sphere: 2/5 m * r**2 -->
```
I have a fork of this repo with all the offending `: `s removed from all the robot urdfs. Luckily, these only show up in urdf comments, so no functionality is changed. I can open a PR, if that's desired?

wxmerkt, 2023-02-15:
Does `rosparam set robot_description -t my_urdf.urdf` work?

james-p-foster, 2023-02-15:
Awesome stuff -- that works. Thanks for your help! I'll have to educate myself on the differences between `rosparam load` and `rosparam set`.

Closing now.

closed 2023-02-15 by james-p-foster