Gepetto/example-robot-data #186: Recover EXAMPLE_ROBOT_DATA_MODEL_DIR in C++

MedericFourmy, 2023-09-27:
I was using the example robot data (4.0.4) in a C++ project by including

```
#include <example-robot-data/path.hpp>
``` 

which was defining the `EXAMPLE_ROBOT_DATA_MODEL_DIR` macro. This include file was deprecated in 4.0.7 and this path seems to be only available as the property INTERFACE_COMPILE_DEFINITIONS of target `example-robot-data::example-robot-data`. 

In my CMakeLists.txt:

```
find_package(example-robot-data REQUIRED)
get_target_property(dir_path_exd example-robot-data::example-robot-data INTERFACE_COMPILE_DEFINITIONS)
message("${dir_path_exd}")
```
prints 
`EXAMPLE_ROBOT_DATA_MODEL_DIR="<absolute_path_to>/share/example-robot-data/robots"`

What's the recommended practice to access it from C++?

nim65s, 2023-09-27:
You can link your C++ project to example-robot-data, eg:
https://github.com/loco-3d/crocoddyl/blob/3de0336178e09b4296912769b494f5741705775f/unittest/CMakeLists.txt#L79

This will define the `EXAMPLE_ROBOT_DATA_MODEL_DIR` macro in the same way, so you can use it with eg.:

nim65s, 2023-09-27:
ref https://github.com/Gepetto/example-robot-data/pull/170

MedericFourmy, 2023-09-27:
Thanks, it worked

closed 2023-09-27 by MedericFourmy