Gepetto/example-robot-data #246: Unitree Go2 joint ordering

EtienneAr, 2024-11-06:
Hello,

When working with Unitree Go2, I noticed an inconsistency between the urdf and the real robot regarding the joint ordering.
The robot would send me it's configuration with the legs in the following order [FR, FL, RR, RL] while the urdf orders the legs [FL, FR, RL, RR]. (the order of the joints per leg match, however)

I was wondering if the URDF should be edited to match the robot ?

([Doc](https://support.unitree.com/home/en/developer/Basic_services) of the robot sdk mentioning in the struct the order of the joint)

cmastalli, 2024-11-06:
Hi @EtienneAr,

The URDF follows the alphanumeric ordering, which, unfortunately, Unitree ignores. Changing the order means, renaming the feet names in such a way that matches the Unitree's order. This as you can see brings other issues.

My suggestion is not to change our example-robot-data URDF. At the end of the day, this package is not intended to be used with our robots.

EtienneAr, 2024-11-07:
Makes sense

Thank you @cmastalli

closed 2024-11-07 by EtienneAr